kicad-spice-library/KiCad-Spice-Library #2: I want to contribute, thinking to include some models..

serveln, 2020-03-21:
Hi! I was looking for a Spice library project especially suited for KiCad. Are you still maintaining this project? If so.. have you seen this page http://ltwiki.org/index.php?title=Components_Library_and_Circuits ? There are a lot of ready available models that could be easily include in your project. Let me know if your are going to continue with this lib!

aster94, 2020-03-21:
Hi! I would be happy to add new modules, the only problem i could see is that someone have to manually download them all, or maybe is there a zip file containing every models?

serveln, 2020-03-21:
There are a couple of zips with everything. I already forked your project, I will upload the new files and then make a pull request so you can review an decide!

closed 2020-08-14 by aster94