kicad-spice-library/KiCad-Spice-Library #16: Number of VDMOS exceeded

mdeweerd, 2021-04-04:
Trying to start a simulation using the Transistor/FET/MOS.lib library results in a crash (even after fixing the first line by changing '#' into '*')

In order to determine the cause, I decided to try to load the library using ngspice.
The behaviour is more or less the same.

File "test.cir":
```cir
.title KiCad schematic
.include "..\KiCad-Spice-Library\Models\Transistor\FET\MOS.lib"
```

Testing file "test.cir":
```
c;\\Spice64\bin\ngspice_con.exe test.cir
```

I performed a "manual binary search" by deleting lines until the library load worked and up to the moment where the load failed.

I then got the following message in the ngspice console (not the KiCAD version):
```
Warning: Syntax check for VDMOS instances is limited to 256 .model cards

Circuit: KiCad schematic
```

This is coherent with the fact that I had to truncate the `MOS.lib` file to 263 lines.

I suggest to split the file into smaller parts.

More generally, all libraries could be checked by trying to load them using ngspice and verify that the ngspice execution does not exit prematurely.

aster94, 2021-04-04:
Hi, I wasn't aware of this limitation, unfortunately I am currently working on other working related projects and these have priority

mdeweerd, 2021-04-04:
That's fine by me - mainly reporting it to potentially help others identify why KiCAD crashes.

holvo, 2021-04-11:
A fix has been uploaded to ngspice git branch pre-master to prevent the crash.
In addition the number of VDMOS models to be checked has been raised to 2048.

aster94, 2021-04-13:
Great!

closed 2021-04-13 by aster94