kicad-spice-library/KiCad-Spice-Library #17: Extract requested models to cope with "Maximum" and "Duplicates"

mdeweerd, 2021-04-04:
In #16 I explained that the MOS.lib library has too many models, which prevents simulation (actually: KiCAD crashes).

I also had a conflict with the 2N7000 model because it appears in Supertex.lib that I loaded fror the VP2106 MOSFET as well as in MOS.lib .
However, one is a "NMOS" model and the otehr one is a "VDMOS" model - each model requiring a different number of nodes (the first model has bulk, the second one doesn't).

So I decided to create a library local to the project (`spice.lib`) where I added a copy of the model.
The benefits are:
- The issue with the limit on the number of VDMOS models is avoided;
- The issue with duplicates is avoided (we know which model is used);
- All the models are included with the project (no need to refer to an external library).

So there would be some use for a script that would extract selected models from the libraries to extract them into a single project spice library file.

Anyway, extracting the model manually is fine for me as well: I do not make that many analogue simulations... .

aster94, 2021-04-04:
> So there would be some use for a script that would extract selected models from the libraries to extract them into a single project spice library file.

This would be a good idea in my opinion, but for the problem I wrote you in the other issue I can't be of much help

Also, I am not very into the KiCad team and they didn't liked the idea of this repository when I posted in the forum (due to problems with copyright) so I think they will stay far from here

mdeweerd, 2021-04-05:
KiCAD does have to care about copyright.  Personnally I think that most manufacturers will be just fine that their models are integrated in any CAD tool because that helps designers to validate their components and design them in.  If it is "too hard" to set up, some will just stick to what is provided.

Put there are some manufacturers that require that you accept some agreement before downloading the models.

Ideally there would be a KiCAD team member that works towards establishing agreements with the manufacturers to integrate their models, and even provide a means for them to suggest updates.

You could add a specific file to list where the original files came from.

mdeweerd, 2021-04-05:
I now have:
```
$ cat doGetSpiceLib.sh
#!/bin/bash
../KiCad-Spice-Library/Scripts/extractModels.pl > spice.lib
cat spice.lib
```

and
```
$ cat spiceModels.lst
2N7000
2N3906
VP2106
SS8050
```

Which results in:
```
$ cat spice.lib
*
* 2N7000 - Extracted from 'Transistor/FET/MOS.lib'
*
.model 2N7000 VDMOS(Rg=3 Vto=1.6 Rd=0 Rs=.75 Rb=.14 Kp=.17 mtriode=1.25 Cgdmax=80p Cgdmin=12p Cgs=50p Cjo=50p Is=.04p mfg=Fairchild Vds=60 Ron=2 Qg=1.5n)
*
* 2N3906 - Extracted from 'Transistor/BJT/BJT.lib'
*
.model 2N3906 PNP(IS=1E-14 VAF=100 BF=200 IKF=0.4 XTB=1.5 BR=4 CJC=4.5E-12 CJE=10E-12 RB=20 RC=0.1 RE=0.1 TR=250E-9 TF=350E-12 ITF=1 VTF=2 XTF=3 Vceo=40 Icrating=200m mfg=Philips)
*
* VP2106 - Extracted from 'uncategorized/spice_complete/supertex.lib'
*
.MODEL VP2106   PMOS (LEVEL=3   RS=0.73            NSUB=5.0E14
+DELTA=0.1      KAPPA=0.0411    TPG=-1             CGDO=6.1716E-10
+RD=1.91        VTO=-2.30       VMAX=3.0E6         ETA=0.12098
+NFS=3.384E11   TOX=1.0E-7      LD=1.6980E-9       UO=865.45
+XJ=6.5E-7      THETA=4.063E-5  CGSO=9.10E-9       L=2.5E-6
+W=0.5E-2)
*
* SS8050 - Extracted from 'uncategorized/Bordodynovs Electronics Lib/cmp/standard.bjt'
*
.MODEL SS8050 NPN IS=377.21E-15 BF=218.082 NF=1.0409 VAF=32.0909 BF=218.08 NF=1.0409 VAF=32.0909 NF=1.0409 VAF=32.0909 VAF=32.091 IKF=.84522 ISE=5.17224E-11 NE=2.12785 BR=5.45795 ISE=51.722E-12 NE=2.12785 BR=5.45795 NE=2.1279 BR=5.45795 BR=5.4580 NR=1.0619 VAR=49.0994 IKR=2.19792 ISC=5.17224E-11 VAR=49.099 IKR=2.19792 ISC=5.17224E-11 IKR=2.1979 ISC=5.17224E-11 ISC=51.722E-12 NC=3.9685 RB=5.74704 IRB=0.1 RBM=0.1 RE=841.72E-6 RC=0.248242 XTB=0.626944 XTI=1 RB=5.7470 IRB=0.1 RBM=0.1 RC=.24824 XTB=0.626944 XTI=1 CJE=27.335E-12 VJE=0.721406 MJE=0.85 VJE=.72141 MJE=0.85 MJE=.85 CJC=26.000E-12 VJC=0.4 MJC=0.367851 XCJC=0.1 VJC=.4 MJC=0.367851 XCJC=0.1 MJC=.36785 XCJC=0.1 FC=.9999 CJS=0 VJS=0.75 MJS=0.5 TF=643.94E-12 XTF=1.43228 VTF=0.344801 ITF=1.20072 XTF=1.4323 VTF=0.344801 ITF=1.20072 VTF=.3448 ITF=1.20072 ITF=1.2007 TR=100.00E-9 PTF=0 KF=0 AF=1 EG=1.0500 CJE=2.73346E-11 VJE=0.721406 MJE=0.85 XTB=.62694 XTI=1 XTI=1
```

mdeweerd, 2021-04-20:
@aster94  I am not sure that this needs to remain open.

aster94, 2021-04-20:
yes we can close

closed 2021-04-20 by aster94