kicad-spice-library/KiCad-Spice-Library #22: Some zener models are unusable

fedorkotov, 2021-08-25:
Some subcirquits in https://github.com/kicad-spice-library/KiCad-Spice-Library/blob/master/Models/Diode/zener.lib library are missing `.ENDS` cards and `DR` models.
For example these ones:
- `DI_BZX84C3V6T`
- `DI_BZX84C3V6TS`
- `DI_BZX84C3V6W`
- `DI_BZX84C3V9S`
- `DI_BZX84C3V9T`
- `DI_BZX84C3V9TS`
- `DI_BZX84C3V9W`

fedorkotov, 2021-08-25:
I think I have found where these models are coming from
https://www.diodes.com/assets/Spice-Models/Discrete-Prodcut-Groups/Zener-Diodes.txt
I will submit a fix sometime later this week.

nexx512, 2021-12-16:
If that's the case then the copyright notice has been removed when importing these model into this library which is a copy right law violation I think.
It says that these model may be coped for research and design and that they have to be copied intact including the copyright notice.
So to avoid any possible legal issues I would suggest to copy that file completely into the repository.

nexx512, 2021-12-16:
However the models in this source file are also missing some `.ENDS` statements. So they might be unusable too.

fedorkotov, 2021-12-16:
@nexx512  That is why I ended up not submitting the fix. Having them all in one place would be very nice indeed but legality of such a compendium is questionable. I'm not a lawyer and can't consult one, so I have to avoid touching this.
It would be nice of course if Diodes Incorporated and others released their models with some more permissive license.

nexx512, 2021-12-16:
@fedorkotov Good point. Unfortunately I already posted a pull request where I manually fixed some models, but only syntactically. They probably don't work for simulation. If i had read this issue before, I wouldn't have done that and stayed away from this repo.