kicad-spice-library/KiCad-Spice-Library #24: Extract Models Scripts Not Working

abust005, 2021-12-04:
[image]

aster94, 2021-12-04:
have you tried with the python script?

denizzzka, 2022-06-12:
@aster94 this is Perl script. I tried to use Perl with path to extractModels.pl script with same result.

It does something as on screenshot above, but doesn't saves any files, at least as I seen

aster94, 2022-06-12:
Yes, I know it's perl, but, I wonder, the same research does it work on python?

gabrielbravov, 2023-08-12:
i have the same problem using Linux Mint. I tried python, python3 and perl
[image]

bringert, 2023-11-09:
extractModels.pl seems to only work for some components, like 2N7000:

```
% perl Scripts/extractModels.pl 2N7000
*
* 2N7000 - Extracted from 'Transistor/FET/MOS.lib'
*
.model 2N7000 VDMOS(Rg=3 Vto=1.6 Rd=0 Rs=.75 Rb=.14 Kp=.17 mtriode=1.25 Cgdmax=80p Cgdmin=12p Cgs=50p Cjo=50p Is=.04p mfg=Fairchild Vds=60 Ron=2 Qg=1.5n)
```

But it doesn't work for others, like PC817:

```
% perl Scripts/extractModels.pl PC817 
*
* PC817 - Extracted from 'uncategorized/Bordodynovs Electronics Lib/sub/PC817.sub'
*
```

I did some basic debugging, and found that this fails, so there is probably something wrong with the regex:

```
   # Extract the model from the file
   if($fileContent=~/$regex/is) {
```

bringert, 2023-11-09:
It looks like the regex can't handle components defined using .subckt, which is a lot of them

bringert, 2023-11-09:
This change seems to fix the problem:

```
diff --git a/Scripts/extractModels.pl b/Scripts/extractModels.pl
index e39bdab..bb0b050 100644
--- a/Scripts/extractModels.pl
+++ b/Scripts/extractModels.pl
@@ -23,7 +23,8 @@ our $continuedNoParenLines='(?:(?<=\n)\+[^\R*()]\R)*'; # Note: regex not validat
 our $modelParenRegex='(?:\.model\s+_NAME_\s+(?<type>\S+)\s*\([^)]*\))';
 our $modelLineRegex='(?:\.model\s+_NAME_\s+(?<type>\S+)\s*[^()]*?\R'.$continuedNoParenLines.')';
 our $modelRegex="(?<model>$modelParenRegex|$modelLineRegex)";
-our $allRegex="(?<matched>${commentsRegex}[ \t]*(?:$modelRegex)\s*?)";
+our $subcktRegex='(?<subckt>^\.subckt\s+_NAME_\s+.*?^\.ends\s+_NAME_\s*)';
+our $allRegex="(?<matched>${commentsRegex}[ \t]*((?:$modelRegex)\s*?)|$subcktRegex)";
 
 # Extract a "model" from a file
 sub extractFromFile {
@@ -43,7 +44,7 @@ sub extractFromFile {
    $regex=~s/_NAME_/$quotedName/g; # Match literal string
 
    # Extract the model from the file
-   if($fileContent=~/$regex/is) {
+   if($fileContent=~/$regex/ism) {
       $model=$+{matched}; 
    }
```

closed 2023-11-09 by aster94