kicad-spice-library/KiCad-Spice-Library #26: Original copyright notices missing and incompatibilities between licences

nexx512, 2021-12-16:
In at least `zener.lib` the copyright notices from https://www.diodes.com/assets/Spice-Models/Discrete-Prodcut-Groups/Zener-Diodes.txt (probably the original source of the file) have been removed.
They should be readded from the original file.
The license of this project might also have to be updated to match the restrictions under which the model providers released the models originally.

aster94, 2021-12-16:
I am not competent in licencing and similar matter. The purpose of this repo was to help people find in a single source a lot of models to ease the spice simulations with kicad.
I am ready to shut this repo down at any moment when requested from anyone of the manufactors or other people

nexx512, 2021-12-16:
I guess that having this repo is not the problem, as long as the sources allow redistribution and are mentioned together with the copyright notices ant their licenses.
One benefit of having this repo is that there is one point where many models can be found and even corrections can be made. I just saw that many of the corrections I made to zener.lib are also present in the original file downloaded from the diodes inc. website.

sixtyfive, 2023-01-06:
You might really want to remove the GPL3 license setting on the repo though. I bet that could get someone angry.

aster94, 2023-01-06:
Probably I will remove the whole repo and organization

sixtyfive, 2023-01-06:
Sad but understandable.

clytle374, 2023-01-07:
You only have the scripts as gpl.  If no one complains, I don't see any reason to remove this.  Just my 2 cents