kicad-spice-library/KiCad-Spice-Library #45: Usage?

lylythechosenone, 2024-12-17:
How is this library supposed to be used? There's no mention in the README, and all issues mentioning usage are instantly closed without much useful information. Is there some way to install it, or should .lib files simply be manually chosen in the file picker each time?

swiwchar, 2025-03-18:
Readme includes information about extracting desired component from the library, and implies there is helpful information there also.

Dinsmoor, 2025-03-24:
I think the dog means "how do I add all of these libraries to KiCad so when I try to run a simulation, the spice files are ready to go"

I am trying to figure out how to add spice files to components in the KiCad schematic right now, and that's how I got here. If you have a resource for that, please let me know. Thanks.